knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #493: [Question] How can I start Auto-Leveling before each print Automatically?

TigerMAST, 2023-10-01:
How to change the start Gcode so that it starts the auto leveling when the printbed is heated and saves the mehs after successful printing?
I use the cura gcode from the manual.

EnableDevice, 2023-10-06:
Hello. In fact, everything is very simple. Command G29 is responsible for automatic leveling. The only thing is that it needs to be started after the bed has warmed up. Therefore, we first turn on the heating, then wait until the temperature reaches the required level, go home (to start counting the coordinates), perform probing of the bed, then turn on the heating of the extruder, wait until it warms up and be sure to turn on the alignment itself. Moreover, alignment can be limited to a certain height of the model (2 mm is indicated here), since this is important precisely on the first layers, and then you can print as usual. In this case, alignment will be performed before each print, and there is no need to save the results. If you want to save the results and simply use them when printing, then perform auto-alignment from the printer menu, and simply specify in the start code "M420 S1 ; Enable leveling". The results will be downloaded automatically.

G21                                        ; metric values 
G90                                        ; absolute positioning 
M82                                        ; set extruder to absolute mode 
M900 K0                                    ; disable lin. adv. if not set in GCODE
M107                                       ; start with the fan off 
M140 S{material_bed_temperature_layer_0}   ; Start heating the bed 
M190 S{material_bed_temperature_layer_0}   ; wait for bed 
G28					   ; Home all axes
G29			                   ; Autolevel
M104 S{material_print_temperature_layer_0} ; start heating the hot end 
M109 S{material_print_temperature_layer_0} ; wait for hotend 
M300 S1000 P500                            ; BEEP heating done 
G28					   ; Home all axes
M420 S1                                    ; Enable leveling 
M420 Z2.0                                  ; Set leveling fading height to 2 mm

TigerMAST, 2023-10-08:
If I leave out M420 Z2.0 then the levling is used for all the printing

EnableDevice, 2023-10-08:
Yes, if you remove this command, the alignment will work across the entire height of the model.

closed 2023-10-08 by TigerMAST