knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #494: [Question] Switching sounds off

stoertie, 2023-10-03:
Hello Oliver,

I installed the "Mega_S_TMC_1.5.2" yesterday and the printer works fine.

But there is a "new" melody when I switch on the printer and when I start (for example) Pronterface or Cura while the printer is switched on and connected via USB.

I find this melody annoying and too loud. I have spent hours making the printer quieter and quieter and then it mutates into a jukebox after a firmware update.... :-)

Is there a way to turn off this noise? 

(The beeps when starting and ending a print job are usually helpful. But these can also be easily turned off with the G-Code M300.

best regards
Michael

knutwurst, 2023-10-03:
Hi Michael,
the startup melody (three different tones) occur when the mainboard ist reset. So every time you turn on or connect to the printer.

If you want to disable it, you have to compile the FW by yourself and disable the startup chime here:

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/e0a4e25ffa59e9e07e5446f7b9e7ca834263f00b/Marlin/Configuration.h#L3171C22-L3172C1

best,
Olli

EnableDevice, 2023-10-04:
Hello.

I compiled a project for you with the sound turned off. There should be no sound when connecting/disconnecting.

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/e0a4e25ffa59e9e07e5446f7b9e7ca834263f00b/Marlin/Configuration.h#L3169-L3171

[MEGA_S_TMC_NO_SOUND.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/12799731/MEGA_S_TMC_NO_SOUND.zip)

stoertie, 2023-10-04:
Hi "EnableDevice",

thank you so much! I will try it today...

Cheers
Michael

EnableDevice, 2023-10-04:
Hi, Michael.

Great! Always happy to help. Write if anything else needs to be changed.

closed 2023-11-01 by knutwurst

stoertie, 2024-01-24:
Hello @EnableDevice ,

I am planning a BL-Touch upgrade of my printer. 

Now I'm afraid that I'll need a different, BLT-compatible firmware version, which would turn my printer into a "jukebox" again. 
Or does the firmware that you kindly changed for me to a "no-sound-fw" already include the BLT function?

In the event that I actually need a "1.5.2-TMC-plus-BLT", could you also create a "NO-SOUND" FW of the BLT variant for me? ;-)

There's no hurry, the BLT sensor is on its way but not yet here and I've been printing without the sensor for about two years, so a few days won't matter. 

Many thanks in advance for your time and help.

Regards
Michael

EnableDevice, 2024-01-24:
Hello @stoertie,

Yes of course without problems. Please note that there are two firmwares in the archive: 10 and 11. This depends on the version of the board. If the logo in the upper left corner directly states that this is “TRIGORILLA 1.1”, then the firmware version for it is 1.1. For everything else, firmware 1.0 is used. In general, if you have any questions, you can always write to me. I also installed BLTouch on Mega S, and if you need help, I will be happy to share what I know. :) You can write by email: [public@enabledevice.ru](url)

[MEGA_S_TMC_BLT_1_5_2_SOUND_OFF.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/14035586/MEGA_S_TMC_BLT_1_5_2_SOUND_OFF.zip)

stoertie, 2024-01-24:
Hi,

thank you so much! ...and many thanks for both versions, I was just about to write an addendum asking if you could edit both versions for me ;-)

Have a nice day, here (in the north of Hamburg/Germany) it's very stormy today, perfect for 3D-printing! :-D

Regards
Michael

EnableDevice, 2024-01-24:
Well, I'm from Russia. Siberia. The harsh region of taiga forests. Blizzards are at the heart of everything here. All that remains is to do all sorts of interesting things at home, in the warmth. :)))

And all the best to you. Write if you need anything.