knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #495: [Question] Reducing firmware size (RAM)

EnableDevice, 2023-10-06:
Hello, Oliver. I still want to enable the "AUTO_BED_LEVELING_UBL" mode (even though it is not fully supported in the printer Anycubic Mega S, it can be used). And I ran into the problem that when I changed the code, there was now not enough memory in the RAM area. What can be safely disabled in this case? A couple of hundred bytes are missing.

knutwurst, 2023-10-06:
Hi @EnableDevice ,
Take a look at #482, where someone has activated InputShaping. Aside from not working well, saving RAM was relatively easy. The information is also in an answer from me in the ticket :)

EnableDevice, 2023-10-06:
Thank you, Oliver!  Have a nice day and thank you very much for your work.

knutwurst, 2023-10-06:
Thank you very much, I wish you the same!