knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #497: [BUG] The target temperature could not be reached

YXG001, 2023-10-21:
### Bug Description

The tool head will not reach the target temperature, and the temperature will stabilize at 10 degrees below the target temperature

### Steps to Reproduce

1. I using octoprint to test it because it will be easer. I set the temperature to 250 degrees, then the octoprint recover:

Recv: ok
[...]
Send: M104 S255

**Expected behavior:** [What you expect to happen]
Generally, the temperature should be 250 degree

**Actual behavior:** [What actually happens]
[image]

2. On this image, you can see the temperature stabilize at 240 degree.

#### Additional Information

If I set to 255 degree, it will stabilize at 245 degree.
But is work well at 210, 220, 230, degree.
It can also reach 250 on the PID autotune.
but if i set the temperature to 230 then 240, it chould reach 240. to get 250 is same.

HaGueRe, 2023-10-21:
A possible cause is the output voltage of the power supply. 
For me it was 9.8 volts without load instead of the required 12 volts and the error pattern was similar. After swapping it for a Meanwell everything was fine again.

kscheff, 2024-01-19:
Since I am struggling a similar heating curve, I suspect too that the power supply is the root cause. Is there any support monitoring voltage, current or power on the Anycubic motherboards? I see that there is the M430 command for doing so, but it is not enabled in the releases:
` M430 - Read the system current, voltage, and power (Requires POWER_MONITOR_CURRENT, POWER_MONITOR_VOLTAGE, or POWER_MONITOR_FIXED_VOLTAGE)`

Anybody experimented with this feature?

coding-pivo, 2024-03-09:
I see a similar issue on my Mega S with knutwurst version 1.5.2. The first layer is printed with required 218C but starting from the second layer the nozzle temperature is lowered by araound 10C. This will not happen with knutwurst version 1.4.2 (maybe 1.4.1). So at least it is not independent from the firmware but maybe a combination of firmware version and power supply?
Also the original firmware is not showing this issue.

closed 2025-02-03 by github-actions[bot]