knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #498: [BUG] 4Max Pro - After use M0, resume is possible only from command prompt

Jaroslaw75, 2023-11-01:
Hallo,

I'm not sure is it a bug or my foult.
I'm using M0 to pause print and insert the nut, after that printer stopping, but resume is possible only from command prompt (if PC is connected I can send M108)
According to Marlin M0 spec. LCD should display message to confirm and unpause machine.

knutwurst, 2023-11-01:
Yes, this is a bug. You can try M600 to pause the print, but it seems that there is an issue with the old Anycubic tft. I'll take a look!

Jaroslaw75, 2023-11-01:
The M600 works, I can resume from the LCD, unfortunately, after using the M600, the printer automatically removes the filament. This is a little bit problematic.

knutwurst, 2023-11-01:
You can use parameters on M600 for custom filament unload/load length. Alternatively you can configure your M600 with M603. See https://marlinfw.org/docs/gcode/M603.html. 

Maybe this could be a workaround for you?

Jaroslaw75, 2023-11-02:
Thanks for the tip, I'll test it.

knutwurst, 2024-06-05:
This is fixed in [Version 1.5.4](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.5.4)

closed 2024-06-05 by knutwurst