knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #500: Modify BLTouch delta on firmware or by Gcode command AND Fw for DRV8825 driver

Sarevok81, 2023-11-06:
### Description

I read on the wiki for the installation of the frimware but I wanted to install the MK4 cart present on thingverse.
This cart has different deltas for the BLTouch position and I would like to know how to change them.

I also have another need, the DRV8825 drivers with TLSmooters are installed in my printer, do I need to change something in the custom firmware to activate these engines?

### Feature Workflow

I followed the instructions for compiling a customized firmware and I managed to compile the MEGA_S_BLT_1.0 FW. However, I would need to know where to modify the positioning deltas of the BLTouch since in the instructions the sensor is installed on the original carriage of the printer and has deltas related to this carriege. Wanting to install the modified MK4 carriage find on thingverse where should I modify the deltas?
Alternatively, are there any Gcode commands to modify these deltas once the firmware has been installed?

I currently have a mega I3 with two Z engines  (no I3 Mega S) on which I modified the drivers and inserted the DRV8825 and TLSmooters. I also installed the Mega S Titan extruder and installed the latest anicubic firmware for the I3 Mega S (I followed the instructions for modifying the extruder directly from anycubic). The DRV8825 drivers require double the steps compared to the originals and the extruder motor requires four times the steps I modified them via GCODE commands and saved them in the EPROM. I then calibrated the steps/mm of the extruder and axes to obtain detailed extrusion and dimensions.
In this situation, is it correct to select the firmware MEGA_S_BLT_1.0?
Do I have to change something in the firmware to be anable the DVR8825?
Where can I find the step/mm settings within the firmware? (I would like to copy the ones currently saved in my EEPROM so as not to have to repeat all the calibrations.)

#### Additional Information

knutwurst, 2023-12-14:
Hi,
To change the offset of the X-Carriage you can simply use the command M851. Just take a look at the instructions. It's described there (relatively far down).

I can't say much about the DRV8825. They should be able to be exchanged 1 to 1 with the normal Pololu A4988. However, I seem to remember that the DRV8825 uses 1/32 microstepping, right? You would have to switch it to 1/16, or double the steps in the firmware. However, you can also do this via gcode. Just enter M92. There you can see the current values of the axes. If your axes only travel half as far as before after installing the DRV8825, then you simply have to double the steps.

Sarevok81, 2023-12-26:
> Hi, To change the offset of the X-Carriage you can simply use the command M851. Just take a look at the instructions. It's described there (relatively far down).
>
> I can't say much about the DRV8825. [...]

thanks, in fact I found the instructions for the delta probe on the wiki.
For drv8825 I have already had to double the steps in the current Fw, so by installing the mega S firmware on tour wiky I will presumably have to repeat the change.

Sarevok81, 2023-12-26:
My mega Is the first model with 220*220 hotbed. I installed the titan extruder upgrade frome mega S upgrade kit and installed the last mega S firmware from anycubic website.
Now is better to use knuctwurst firware for mega or for mega S?
In the firmware there Is a bed dimensioni information?

closed 2024-11-21 by github-actions[bot]