knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #501: [BUG] Filament Runout Sensor - Random False Positives

spaelectronics, 2023-11-27:
I keep getting random false positives on the Filament Runout Sensor.

I thought it was the switch went bad, so I replaced the switch, but I'm still having the same issue.

fatalyzm, 2023-11-30:
Same here, the filament sensor runs me crazy.
Unfortunately i have to disable it statically.

I'm still on v1.5.2 - because there are n signs for a fix.

Anycubic Mega X DGUS TMC
Trigorilla 0.0.2

Additionally the Special Menu disappears after a few minutes.

elodur, 2024-01-02:
Same problem, but I'm not Sure if it's a software bug. When I bridge the switch with a wire there is no problem any more. Maybe it just is the cheap switch.

stklcode, 2024-01-02:
I can also confirm this, printing occasionally stops with runout warning. Unfortunately, I didn’t investigate yet, but I would also suspect the hardware.

One idea would be tweaking the runout settings, e.g. enable soft-pullup/down, if there is no hardware equivalent or extended the debounce logic programmatically.

Configuring `FILAMENT_RUNOUT_DISTANCE_MM` could also make sense, there are at least 50-100mm that can safely be ignored, before the extruder fails in default setup. I didn’t check the internal logic, if the state is re-checked in the loop, even a small value may sufficient to debounce a faulty switch.

spaelectronics, 2024-01-03:
> I can also confirm this, printing occasionally stops with runout warning. Unfortunately, I didn’t investigate yet, but I would also suspect the hardware.

I thought it was a faulty switch too, so I replaced the switch with a new one, however I still have the same issue.

> One idea would be tweaking the runout settings, e.g. enable soft-pullup/down, if there is no hardware equivalent or extended the debounce logic programmatically.

I too was thinking it could be a pullup/down issue. I haven't tried it yet, as I can't compile it myself as I don't know the proper configuration settings for the Mega X.

stklcode, 2024-01-06:
Just compiled a FW with `#define FIL_RUNOUT_PULLUP`.

Doesn't seem to make much of a difference, 2 false positives before leveling started and one in the second layer... Disabled the sensor for the current print job. :disappointed: 

Next I will try to extend the default debounce threshold from 5 to something higher like ... maybe 10 or 20.
```c
#define FILAMENT_RUNOUT_THRESHOLD 5
```

If that also does not help, I'll switch from debounced to delayed trigger by setting the distance to something >0, like
```c
#define FILAMENT_RUNOUT_DISTANCE_MM 25
```

stklcode, 2024-01-07:
I did a second test with
```
#define FILAMENT_RUNOUT_THRESHOLD 20
```

No false-positive runout trigger during a 3h print job :tada: 

When I pull he sensor cable, it claims runout. Little sluggish, but the feature still works and we have at least 100mm buffer, before the extruder runs empy on default setup, so I'd say that's fine.

PR #506 (2024-01-07): increase threshold for filament runout
### Description

We have noticed regular false-positive filament runout alarms on various printers.

Increase the threshold from 5 to 20 to reduce the chance of false positive triggers.

Build and run tested on an Anycubic i3 Mega P.

### Requirements

--

### Benefits

Reduce the risk of false-positive filament runout alarms. At best there is no issue anymore, at the very least the number should be reduced. We have to try out...

### Configurations

--

### Related Issues

resolves #501
closed 2024-01-07 by knutwurst