knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #503: [BUG] (1.5.2 Ignores limit switches with print start)

macchinette, 2023-12-04:
### Bug Description

Hi Oliver,
I have a MEGA S to which I mounted the TMC2209 drivers and until yesterday I never had problems with your firmware with the MEGA_S_TMC versions up to version 1.4.3 then I switched to the latest 1.5.2 and I had a problem launching a press that ignored the limit switches.

I don't use MESH and by changing the complete hotend to change the nozzle diameter I adjusted the offset from the software (Superslicer) for the various hotends.

My start G-code is written like this:
G90 ; use absolute coordinates
M83 ; extruder related modes
M140 S[first_layer_bed_temperature] ; set bed temp
G28 X0 Y0 ; move X/Y to min endstops
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; set extruder temp
M109 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; wait for extruder temp
G28 Z0 ; move Z to min endstops
G1 Z0.46 F360 ; raise nozzle 0.16mm
G1 X2 Y2 F1200 ; move X/Y to prep for purge
G92 E0 ; zero the extruded length
G1 X80 E29 F720 ; initial load
G92 E0 ; zero the extruded length
G1 X160 E29 F720 ; intro line
G92 E0 ; zero the extruded length

I wonder what's missing because he ignored the Z limit switches?

The only difference is that I had made a MESH just to try and it wasn't my intention to use it because I don't need it.

Then I went back to firmware 1.4.4 and there I also noticed that by doing the MESH but not using it the level of the platter was impossible to do, at that point I reset to the default values and I managed to level the platter and return to the condition original.

Is it possible that launching a print will load the MESH if present even without M420 S1 in the Start G-code, while if you go HOME from the display the Z limit switches are used?

Thanks for your help.
Macchi

knutwurst, 2023-12-14:
I'm not sure I understood the problem.

The limit switches are only read once. Namely when G28 (home) is running. They are then ignored and replaced by software endstops.

If you replace the pressure plate, you will of course have to rehome and either replace your mesh or recalibrate it. Or you deactivate the mesh.

macchinette, 2023-12-14:
> I'm not sure I understood the problem.
>
> The limit switches are only read once. Namely when G28 (home) is running. They are then ignored and replaced by software endstops.
>
> If you replace the [...]

Thanks for the reply.
So I think that during the holidays I will try the MEGA_S_TMC_v1.5.2.hex version again and in the start G-code I will insert M420 S0 to disable the use of the mesh of the G28 Z0

Then we'll see if it works.
Thank you

closed 2024-06-05 by knutwurst