knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #504: [BUG] (short description)

Druckmichel, 2023-12-14:
Hello,
I installed an BLtouch on Anycubic i3 Mega Pro following your Beginner's Guide. 
Flashed "MEGA_P_DGUS_BLT_10_v1.5.2.hex".
Via Octoprint I did Determination and configuration of the Z offset, so far no issue.

When it gets to G29/ Meshleveling the Printhead lift up and travel to point one and starts probing midair.
(Depoying and retracting the pin. No Z travel downwards touching the bed.)

knutwurst, 2023-12-14:
This is most likely due to incorrect wiring of the sensor. It triggers even though it doesn't trigger. White and black swapped? ;)

Druckmichel, 2023-12-14:
Wow response time Zero :)

I swapped it for testing but then it's not working. I was wondered about that behavior because "Determination and configuration of the Z offset" went very well.
Wiring double checked five times.

knutwurst, 2023-12-14:
Take a look at your hotend board. Are there various small SMD components installed there? Maybe it's because. However, setting the Z offset might not work otherwise. After all, you have to touch the bed with the sensor once. I still suspect an electrical problem. There's a loose connection somewhere or a cold solder joint or something.

Druckmichel, 2023-12-14:
You are so right.
It was my fault, forgot swapping A5 / B5 on the green connector.

Sent you Coffee on PayPal.
Merry Christmas from East Frisia

closed 2023-12-14 by Druckmichel

knutwurst, 2023-12-14:
Never mind. Everybody makes mistakes. I destroyed the stepper drivers the first time. Or where do you think the images in the wiki come from? ;)

I'm glad it works now!

Thank you and Merry Christmas! <3