knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #507: [BUG] Increase steppers deactivation timeout

enaix, 2024-01-07:
### Bug Description

With current Marlin configuration, the stepper drivers will deactivate after 1 minute of inactivity

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/f9164e3a6e0f859303c73f2492404a32e1c8f05f/Marlin/Configuration_adv.h#L1120

Perhaps setting this value to 20 minutes would work (`#define DEFAULT_STEPPER_DEACTIVE_TIME (20*60)`)

### Steps to Reproduce

Wait 1 minute after performing any command

**Expected behavior:** The axes wouldn't deactivate so quickly

**Actual behavior:** It deactivates

stklcode, 2024-01-07:
What would be the benefit of 20min default timeout? A minute or 2 sound like a reasonable default to me.

You can always override the default via G-Code M18 [1], if you need to keep the steppers active for a specific use case.

[1] https://marlinfw.org/docs/gcode/M018.html

enaix, 2024-01-07:
I see that it's unimportant, it just differs from the stock firmware and can be confusing during manual bed leveing. The user may accidentally push the axes without noticing and the coordinates would become out of date

stklcode, 2024-01-08:
Hm, think I never ran into this, but it’s a good point.
Sounds like the manual leveling feature should disable or increase the timeout, so the axes are static until the process is done.

enaix, 2024-01-09:
The problem is that the behavior differs from the default firmware: if the user tries to do the leveling the exact same way as on the old firmware, they may not activate 4 point leveling option in the special menu - at least this was in my case :)

stklcode, 2024-01-09:
Please define “default“. Unmodified Marlin or the Anycubic Firmware (based on Marlin 1.x)?

There are a lot of differences from the Anycubic FW, I would not call the fact that is it different a “problem“ - that’s why we are using the custom build and don’t stay with the stock one.
Does the Anycubic FW contain any kind of auto-off at all (didn’t use it for years, so I cannot tell)? If not, I’d call this an improvement.

I never used the 4-point leveling feature from the special menu, just occasionally from the built-in menu of the display (Setup > Leveling). It just moves the nozzle to 4 pre-defined points when pushing “next“ so nothing special about it, or is there something else I am missing?

So, if you start leveling, no matter if you use the special menu or the build-in menu, extending or disabling the auto-off timeout when moving to the first point would solve the issue of accidental movement, doesn’t it?

enaix, 2024-01-09:
I mean the anycubic firmware. Of course the feature itself isn't the problem, I misunderstood what "manual leveling feature" means in the firmware. (I've meant that the short timeout not being mentioned anywhere may cause confusion).

So, if I understood correctly, if the user manually moves the axes (G0/G1), then we should perform the check for the corner points (with some margin to account for the bed clips). This for sure will work, I haven't thought about the feature this way. I may create a PR, but it may take some time to get used to the code guidelines.

closed 2024-06-02 by knutwurst