knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #511: i3 Mega S with BL-Touch temporary nozzle height problem

stoertie, 2024-02-06:
Hi all,

I am not very familiar with Github and hope you will forgive me if I ask my question here. I have not found out how to set the "help wanted" label. 

Since yesterday I have a BL-Touch sensor on my i3 Mega S. The firmware I have installed is MEGA_S_TMC_BLT_1_5_2, in which "EnableDevice", has switched off the melodies again, as in the version without BLT befor. (Thank you very much!)

The entire set-up process went perfectly, the automatic levelling works fine and the printer also prints (mostly) without any problems. 
Yesterday, however, I suddenly had the problem that the nozzle was too low above the print bed and the material could hardly be pushed out. Luckily I was sitting next to it and was able to cancel the print. I then did a new auto levelling, restarted the print and everything went fine.

This morning I ran a longer print... Everything worked perfect. After I started the second printout, the print head went so deep into the print bed that the printer just chattered. (a terrible noise!)
I switched the printer off and on, did a new auto levelling and now it prints again without any problems and the first layer was also perfect.

I use Cura 5.5.0 as a slicer and have connected the printer via USB.

What could be the reason for the print head sometimes being positioned too low? Does anyone have any ideas? 

Best regards
Michael

alexaring, 2024-03-03:
I can confirm that I have currently similar issues, you are not alone. I suspect that "sometimes" the z endstops are not working. I will going back to 1.4.4 and try to figure out if the problem is gone and report back.

That's all what I can say to this and yes it is a terrible noise... luckily it is at in my case when I am homing. Sometimes it doesn't happen especially when I do auto-leveling before. I saw some other open issues that sound similar.

stoertie, 2024-03-04:
Hi Alex,

yes it's really strange. 
I have had this issue only before I posted my problem. 
Since I posted this issue everything works well. Also after doing the auto-leveling several times everything works perfect. 
So, for me and at the moment, there is no reason to go back to the previous firmware.

Regards
Michael

closed 2025-01-30 by github-actions[bot]