knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #512: [HELP] bltouch probe won't extend during auto leveling

RemagOfficial, 2024-02-11:
I recently installed a bltouch on my anycubic mega x by following this repo but when I try and run the auto leveling option the probe doesn't extend. it does extend twice on startup and the red led is lit continuously as I've seen in people's videos about installing one, I've checked my wiring and tried both the _10 and _11 firmwares (v0.0.2 main board and server header in S1) and neither changed anything so I have no idea how to proceed since I've followed everything correctly and it's not working.

RemagOfficial, 2024-03-01:
can someone please help, I've had this printer for over a year now and haven't been able to make a good print because of a bunch of issues, the bltouch is the last thing I need to do to be able to use the printer

KleinerKobold, 2024-03-14:
I have a similar problem with my bl touch, it already worked, and now when I start auto level the head homes and drives to point 1 then when the probe should extends nothing happends and after a few moments it peeps and end.

knutwurst, 2024-03-14:
I'm on it :) 
Sorry for the delay... I'll keep you updated

RemagOfficial, 2024-03-15:
the only thing in the faq that seems to be about my issue says that the black and white wires could be swapped, I'll check to be 100% sure but it should be correct

RemagOfficial, 2024-04-27:
hello?

closed 2025-03-24 by github-actions[bot]