knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #514: [BUG] "Print Completed" not poping up after print has finished

phenolplus, 2024-02-16:
### Bug Description

When a print is complete, a popup showing "Print Completed [time] ... [OK]" would come up.
Some times it doesn't.
The printer will be stuck in printing mode, showing progress 100%.
Because of this, you cannot move axes, filament, or start another print.
If you click [Pause], the printer errors out. (failed to get response from pause command)
If you click [stop], the pop up appears and the print is completed.

Note: The g-code file is indeed complete because the last line was executed (M104 S0).
I notice this only happens on consecutive prints. It never happens on a first print after power on.

### Steps to Reproduce
0. Print from SD card from root folder.
1. Start a print.
2. Wait till it completes. Then start another print
3. Wait till it finishes at 100% and stop.

**Expected behavior:** [What you expect to happen]
The print finishes and a popup occurs in the screen.

**Actual behavior:** [What actually happens]
The print stays at 100% progress.
The printer does not respond to [pause] button.

#### Additional Information
This can be tested with single line g-code file.
It never happens on first print after power up. However, sometimes it doesn't happen on a second print either.

Machine: stock i3 mega-s
Firmware: MEGA_S_v1.5.2.hex (prebuilt release)

potatobarry, 2024-02-29:
Could reproduce, same Firmware, same Printer.

knutwurst, 2024-03-11:
That's indeed a bug. I'll take a look. Thanks for reporting!

phenolplus, 2024-10-08:
Following up with what I think may be the same issue/cause. When filament runs out at 99% progress it will hang. The printer has responded to the no-filament error (by pulling back filament, homing print head, and turning off heater) but there's no pop up "Lack of filament [OK]" warning. The printer instead becomes unresponsive (pause fails). The print is not recoverable.

closed 2025-09-05 by github-actions[bot]