knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #516: [FR] Can you pls make a Firmware Version fou the Mega X DGUS TMC with Inverted Extruder Motor

Storma-X, 2024-03-07:
Hey, can you make a Firmware for the Mega X with the DGUS Display and the TMC Motors with Inverted Extruder Motor? 

Just like the Normal MEGA_X_DGUS_TMC but only with an inverted Extruder Motor?

knutwurst, 2024-03-11:
Hey @Storma-X ,
uh... no :) 
You can just invert the extruder step value with the M92 gcode.

`M92 E-396`

Otherwise you can compile the firmware by yourself. See: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Compile-Firmware-with-PlatformIO

closed 2024-03-11 by knutwurst

Storma-X, 2024-03-15:
Hey, where can i change the Value "M92 E-396" in my normal gcode is no Line with M92

SirMike3D, 2024-06-16:
Hi!
When compiling an own version with inverted extruder, which file(s) and which line(s) have to be modified?
What are the E-Steps for the MEGA-S extruder? I am asking because I have updated my MEGA 1G with the MEGA S extruder upgrade kit and I do not know, if the MEGA S extruder has the same e steps like the old (1G) one.
BR Mike

stklcode, 2024-06-16:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/1ef65dc26ba2d697c6ed1dc0eddc5fbb219d964f/Marlin/Configuration.h#L1378-L1384

89 for the Mega 1G vs.393 for the Mega S