knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #518: [BUG] Mega Pro very loud Z-Stepper

CraftoHohenvels, 2024-03-16:
### Bug Description
My Mega Pro's z-Stepper is really loud when this fw is installed. I can't tell what causes this exactly, but it is this fw for sure. (Before installing it it was silent and after flashing the original fw it is silent again)

### Steps to Reproduce
1. Download the [firmware](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.5.2/MEGA_P_DGUS_TMC_v1.5.2.hex)
2. Flash the firmware using OctoPrint Firmware Flasher
3. Move the head on the z-Axis

**Expected behavior:** sound silent as before and after

**Actual behavior:** The steppers sound a lot louder

stklcode, 2024-03-16:
You link the TMC flavor of the firmware and compare against “original“... 🤔 Did you actually replaced the stepper drivers?

CraftoHohenvels, 2024-03-16:
My bad, I linked the wrong fw, it is actually the one without TMC

knutwurst, 2024-06-02:
Since the Mega Pro only has the Trinamic stepper drivers on X and Y, the Z axis is of course louder.

However, the stock firmware does not use mesh bed leveling, which is why the Z axis is normally completely immobile. If you now use mesh bed leveling, the Z axis always works and causes noise ;)

closed 2024-06-05 by knutwurst