knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #519: [BUG] Mega Pro built-in bed-leveling no longer working

CraftoHohenvels, 2024-03-16:
### Bug Description
With this fw my Mega Pro will home when using the vanilla bed leveling. But if you hit next it doesn't react. Like nothing, except the stepper making noises. It works after reflashing the original fw.

### Steps to Reproduce
1. Download the [firmware](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.5.2/MEGA_P_DGUS_TMC_v1.5.2.hex)
2. Flash the firmware using OctoPrint Firmware Flasher
3. Try to use vanilla leveling

**Expected behavior:** move to the next position

**Actual behavior:** The steppers just product very silent noises, but the head doesn't move.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

stklcode, 2024-03-16:
Already fixed in #509 (not released yet)

CraftoHohenvels, 2024-03-16:
Good to know!

closed 2024-03-16 by CraftoHohenvels