knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #521: [BUG] New version new problems

CraftoHohenvels, 2024-03-24:
### Bug Description

When starting a print the printer homes, which is common behavior. So when starting a print (or basically just homing) the printer disconnects upon finished homing.

This is on my AMP.

### Steps to Reproduce

1. Install version 1.5.3 (proper version for the configuration)
2. Flash it via Octoprint l
3. Start a print or home

**Expected behavior:** complete the homing an move on

**Actual behavior:** complete the homing and disconnect

knutwurst, 2024-06-02:
Does this behavior also happen if you install an older firmware? To me it sounds like a hardware error. Possibly in the heating cartridge.

CraftoHohenvels, 2024-06-14:
Sorry for the late response!

It only happened with the version 1.5.3, earlier versions are not problem - same goes for the "vanilla" firmware. I haven't tried version 1.5.4 yet.

I would rule out a hardware problem

knutwurst, 2024-06-14:
That's a good start. Please try 1.5.4, as it fixes a few bugs that affect the control of the end stops and motors.

Otherwise, I would like to ask you to write down exactly what kind of printer you have and what has been changed. I will then try to reproduce it and find the bug.

closed 2025-05-11 by github-actions[bot]