knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #523: Fährt Z Home nicht an

Leon751, 2024-03-30:
Hallo,
ich habe die Version für MEGA S_TMC über VC Code Marlin Auto Build kompiliert und geflasht.
Dazu habe ich in der INI all die anderen Versionen ausgeklammert, weil er ja sonst alle kompiliert und das natürlich dauert.
In der Config.h habe ich nur die PID Werte für das Hotend und Bett eingetragen, sonst ist alles im Ursprungszustand. 

 #MEGA_TMC_BLT_10
  #MEGA_TMC_BLT_11
  #MEGA_S
  MEGA_S_TMC
  #MEGA_S_BLT_10
  #MEGA_S_BLT_11
  #MEGA_S_TMC_BLT_10

Ich kann alle Achsen ein Stück verfahren, Home geht auch für X und Y, jedoch wenn er Z Homen soll tut sich nix.
Die Achse bewegt sich kein Stück. Ich muß die beiden Z Endschalter von Hand drücken damit er andere Befehle wieder ausführt.

Verbaut habe ich die TMC2209 auf allen Achsen.
Dieses Setup läuft mit der Version MEGA_S_TMC_v1.5.3.hex, deshalb schließe ich darauf das es hier um ein Bug 
Vielleicht mache ich aber auch einen Fehler?

Danke schon mal für Ihre Hilfe.

knutwurst, 2024-03-30:
Bitte lies einmal die Anleitung zum Kompilieren :) 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren

Leon751, 2024-03-30:
Erstmal danke für die schnelle Antwort.
Also, habe mir die Anleitung durchgelesen.
Habe es auch neu entpackt und nach der Anleitung kompiliert und geflasht.
Die Version 1.53 funktioniert ja, das hatte ich auch schon als Hex Datei geladen und geflasht. 
Die ist allerdings ohne Input Shaper.
Die Version mit Input Shaper dagegen funktioniert nicht. Da ist das oben beschriebene Problem mit dem Z Homen.
Hab aber auch gelesen das die Master Version nicht unbedingt Fehlerfrei ist und wohl noch dran gearbeitet wird.

knutwurst, 2024-03-30:
Gerne :)
Input Shaping wird schon von der Hardware so nicht unterstützt - jedenfalls nicht sinnvoll. Es gibt dazu schon diverse geschlossene Tickets wie dieses hier: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/402

Und ja, der Master sollte funktionieren, aber muss es nicht. Dort sind momentan die kompletten Marlin Updates drin und das ist noch nicht getestet.

stklcode, 2024-03-30:
> In der Config.h habe ich nur die PID Werte für das Hotend und Bett eingetragen, sonst ist alles im Ursprungszustand.
...
> Die Version mit Input Shaper ...

Input Shaping ist schon eine vergleichsweise große Änderung und nicht wirklich "Ursprungszustand" :thinking: 

Btw. auf welchem Stand setzt du auf, _1.5.3_ oder _master_? Dazwischen liegt auch schon ein "kleiner" Unterschied.

Fang doch erst einmal von vorne an: Firmware auf dem Tag v1.5.3 ohne Änderungen zu kompilieren. Wenn das funktioniert, kann man mit Modifikationen beginnen. Nicht direkt 3 Dinge anfassen und dann rätselraten, was der Grund für den Fehler ist.

----

Das eingangs beschriebene Problem sieht mir nach fälschlicher Weise invertierten Z Endstops aus, d.h. der Drucker "denkt", dass er ganz unten ist und verfährt entsprechend nicht abwärts, bis du die Schalter drückst. 

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/1.5.3/Marlin/Configuration.h#L1251

stklcode, 2024-04-30:
Dürfte einer der (vergleichsweise vielen) Bugs in 2.1.2.2 sein:
https://github.com/MarlinFirmware/Marlin/issues/27014

stklcode, 2024-06-05:
Sollte mit dem verlinkten PR #537 gelöst sein.

Eine Zeile aus dem aktuellen Entwicklungszweig fehlte. Eigentlich logisch, dass es irgendwo in dem Umfeld sein musste, aber offensichtlich ist der Fehler nicht ... insbesondere da es wohl Personen gibt, die es durch reine Pin-Konfiguration behoben haben.

Lorbeeren gehören nicht mir, hab zwar mit diskutiert und einiges über die ganze Nachverarbeitung der Pins gelernt, aber am Ende hat Peter die Lösung gefunden (zuvor verlinktes Marlin issue hat die ganze Diskussion)

PR #537 (2024-06-05): fix STEPTEST macro to restore homing with dual Z endstops
### Description

This macro builds a string starting with `HAS_` but `HAS_` was replaced with `USE_` in Marlin 2.1.2.2.

Backport the upstream patch should restore correct endstop polling and fix the bad homing behavior.

### Requirements

any target with dual Z endstop

### Benefits

Z axis homing works again.

### Configurations

--

### Related Issues

fixes #523
commit 5eac0f64 (2024-06-05): fix STEPTEST macro to restore homing with dual Z endstops

This macro builds a string starting with "HAS_" but "HAS_" was replaced
with "USE_" in Marlin 2.1.2.2. Backport the upstream patch should
restore correct endstop polling and fix the bad homing behavior.

Fixes: #523
closed 2024-06-05 by knutwurst