knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #525: [BUG] Sometimes the printer hangs

anno0, 2024-04-01:
### Bug Description

I print from SD card. Sometimes the printer hangs. It beeps a few times in a higher frequency and the print head goes up. This happens if I want to start a new print.

### Steps to Reproduce
1. The printer cools down from the last print. The bed temperature is 40 °C and the print head temperature > 50 °C. So the fan is running.
2. I have selected the print menu and have selected a model to print.
3. Start the print
4. After a few seconds it beeps in a higher frequency, the print head goes up and holds it current temperature. Well this is displayed in the displayed.

**Expected behavior:** I would expect that the print is started without error

**Actual behavior:** Sometimes this bug effects that I have to power off/on the printer.

khaninD, 2024-06-02:
Hi, reproduced on i3 mega s with TMC too!( sometimes printer is freezing after started a few minutes of print. Sometimes i catch this error on 90% of printing.

knutwurst, 2024-06-02:
Can you post a log output of your printer? Please also read the [FAQ](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(english)). There may already be answers to these questions.

closed 2025-04-30 by github-actions[bot]