knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #526: [BUG] Anycubic Mega Pro keeps printing on the front right corner

Blubberland, 2024-04-03:
## Bug Description

Anycubic Mega Pro keeps printing on the front right corner, instead of the middle of the bed.

### Steps to Reproduce
The printer hasn't been modified.

Following this guide step by step to step 8(*):
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)
- Identidied Printer: Mega Pro (Mega S with laser, BMG extruder and TMC drivers on X and Y. The Anycubic 0.0.2 display is always used)
- identified display: 2.3 The Anycubic 0.0.2 display (DGUS clone)
- Identified Mainboard: Trigorilla 0.0.2
- Chose Printer in Cura: Anycubic i3 Mega S/Pro
- Downloaded Firmware: MEGA_P_DGUS_v1.5.3.hex (also tried v1.5.1 because someone had a problem with a newer version. The problem was unrelated to mine, but tried nonetheless)
- Installed the FW via Cura
- Loaded _default settings_ and _saved EEPROM_
- didn't changed the bed size (after trying it out and revoked later, because it printed over the edge)
- ticked _Origin at center_ (after trying it unticked)
- Adjusted _Mesh Fixes_ and _Cooling_ in Cura
- Used recommended Start and Endcode (also used the default which I assume is provided by the profile in Cura - no difference regarding this issue, which I assumed beforehand, without knowing anything about gcode)

**Expected behavior:**
Print the object in the middle of the bed

**Actual behavior:**
See bug description

## Additional Information

### Board
[image]

### Results
Pictures and video was made without changing the bed size, because otherwise it was printed over the edge
[image]
[image]
[image]

Blubberland, 2024-04-03:
When using Octoprint, the following error appears, which looks like it was related to the issue:
[image]

I get the feeling I'm doing something wrong in Cura...

stklcode, 2024-04-03:
> ticked Origin at center (after trying it unticked)

The actual origin is in the front left corner, not in the center.
Your object is sliced to be printed at -11mm in X/Y direction. It’s not possible to print outside the bed and you are running into (soft) endstops.

Blubberland, 2024-04-03:
Yep, figured that. Thank you very much anyway!
Now I have to figure out why Octoprint throws Autoleveling-Errors.

closed 2024-04-03 by Blubberland