knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #527: Firmware version issue

PNetrox, 2024-04-04:
Hi, i have problem choosing correct firmware.
My printer is Anycubic 4Max Pro (not v2.0):
[image]
All i know is that i don't have BL-Touch and I have original DWIN / DGUS II display. But i don't know if i have a TMC.
So, how can i check it and choose right firmware?

stklcode, 2024-04-04:
> But i don't know if i have a TMC.

Did you install TMC drivers by yourself (or got a used printer with modified drivers)? If not, the answer is most likely “no“, factory setup does not come with updated drivers.

Anyway you should notice the wrong FW immediately, as the axes will move in the wrong direction. If you flash the non-TMC version, push “home“ and printer moves to all zero without strange noises, everything is fine.
If it works with unmodified original FW this should be the case.

If you wanna be 100% sure you can disassemble the printer, look at the board and check which drivers are installed there.

Nebuli1, 2024-04-04:
4MAXP2_v1.5.3.hex will be a suitable file ? Will it be a good file for the first edition of 4Max Pro? I don't see P1 for the first pro version.

Nebuli1, 2024-04-11:
@stklcode

stklcode, 2024-04-11:
Quoting v1.4.0 Changelog:
> *4MAX Pro and 4MAX Pro v2 share the same firmware. Please use the 4MAXP2 binary suitable for your Display (Original DWIN II or Anycubic 0.0.2)

My guess would be the `4MAXP2_DGUS` flavor. But unfortunately. I have no idea if this is correct as I’m not familiar with the 4Max series.

knutwurst, 2024-06-05:
I don't have a 4MAX Pro myself, but I couldn't see any differences from the original Anycubic firmware except for the new extruder. So the V2 firmware should work.

JohnEdwa, 2024-09-17:
AFAIK the only relevant difference is that the 4Max Pro has a build volume of 270mmx205x205mm, while the 2.0 has 270x210x190 because it has the [bearings centered on the bed frame](https://www.lesimprimantes3d.fr/wp-content/uploads/2020/10/Anycubic-4Max-Pro2-axe-Z.jpg) so they don't collide with the print head (+5mm Y) but do hit the new beefier mounting plate at the bottom (-15mm Z)

closed 2025-08-16 by github-actions[bot]