knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #534: Double extruder on single hotend

AleScrMaker, 2024-05-21:
I wish to personalize your source code adding the second extruder but I have only one hotend. I have a mega-x and I want to use the bltouch in future with your customized menu. how can I customize the source code?

knutwurst, 2024-06-02:
Please take a look at the [wiki](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki). There you will find instructions on how to modify and compile the source code yourself.

AleScrMaker, 2024-06-02:
Thank you to answered me. I’ve already read the wiki page and compile marlin but I’ve problem with z homing. When I start homing with g28 the printer don’t moving and in console appear echo:busy ecc. In order to move the axis it’s necessary restart the printer. I don’t understand why.
In last message I do a mistake. My printer is a mega s

stklcode, 2024-06-02:
> When I start homing with g28 the printer don’t moving and in console appear echo:busy ecc. In order to move the axis it’s necessary restart the printer. I don’t understand why. In last message I do a mistake. My printer is a mega s

Z axis leveling is broken on current _master_ branch, see #523 (German) or upstream issue https://github.com/MarlinFirmware/Marlin/issues/27014
Try starting from the [1.5.3](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/tree/1.5.3) for a codebase which is known to be working.

AleScrMaker, 2024-06-02:
Ok, I'll try tomorrow. I have another question about the second extruder. I connected it to the motherboard using the e1 connector (originally anycubic connected the second z stepper on the e1 connector of the motherboard). which pin should I set in the source code?

AleScrMaker, 2024-06-03:
hi, i tried the 1.5.3 release and the z axis move in the right direction but now the endstop doesn't work. Only x axis endstop works, the y and z axis move in the right direction but when the endstops are triggered the stepper continues to move and I'm forced to shut-down or reset the printer.

closed 2024-06-11 by AleScrMaker