knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #535: [NEED HELP] Heating Failure: Stuck at Specific Temperature

rlongdragon, 2024-05-23:
### Bug Description

I am using an Anycubic Mega S. When using the MEGA_S_BLT_10_v1.5.3 and MEGA_S_BLT_10_v1.5.0 firmware versions, I set the hotend temperature to 230°C. Initially, the temperature rises steadily until it approaches 225°C. After exceeding 225°C, the temperature curve flattens out and eventually gets stuck at 228°C. Although it shows 228°C, the heating has actually stopped, and the hotend starts cooling down, but the detected temperature remains at 228°C. It only shows the correct temperature after I restart the machine.

### Steps to Reproduce

1. Use firmware version MEGA_S_BLT_10_v1.5.3 or MEGA_S_BLT_10_v1.5.0
2. Set the hotend temperature to 230°C
3. Observe the temperature rise to 225°C and then flatten out, getting stuck at 228°C
4. Notice that the actual temperature of the hotend is dropping even though the display shows 228°C
5. Restart the machine to see the correct temperature reading

**Expected behavior:** The hotend should reach and maintain the set temperature of 230°C.

**Actual behavior:** The hotend temperature gets stuck at 228°C on the display, while the actual hotend cools down.

#### Additional Information

* This issue has been reproduced many times and always gets stuck at 228°C.
* This problem does not occur with firmware versions MEGA_S_BLT_10_v1.3.0, MEGA_S_BLT_10_v1.3.1, or the official firmware for the 3D printer.
* I am not a native English speaker, and I used AI for translation.

rlongdragon, 2024-05-28:
I'm not sure what I did that allowed it to heat successfully. It might have been reinstalling the firmware.

However, there are still many issues, such as the one mentioned in the FAQ:
[My printer stops when printing starts. When I run Pronterface/OctoPrint and look at the console output, it says: Error: XXXXX Failed. System stopped. Heater_ID: X. Error: Printer halted. KILL called. That was not so before.](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(english)#my-printer-stops-when-printing-starts-when-i-run-pronterface--octoprint-and-look-at-the-console-output-it-says-error-xxxxxx-failed-system-stopped-heater_id-x-error-printer-halted-kill-called-that-was-not-so-before)

But when the error occurs, although the displayed temperature is 210°C (I set it to 235°C), the actual temperature keeps rising, and the cooling fan does not activate. By the time I notice and restart the machine, the temperature has already risen to 280°C.

knutwurst, 2024-06-02:
In most cases, this is due to a faulty thermistor, a misaligned component fan, or incorrect PID values.

Can you successfully perform PID tuning? You may need to do it for higher temperatures (namely the ones you are using). If the PID values ​​are incorrect, the temperature may not be reached, or may overshoot the target.

rlongdragon, 2024-06-09:
I used the command M303 C8 S175 and it successfully resolved the issue. The printer now appears to heat correctly and maintain a stable temperature during printing.

knutwurst, 2024-06-09:
I don't quite understand why the target temperature of 175°C works for your problem, since you should actually specify the temperature that you want to keep stable (in your case 230°C). Nevertheless, I'm happy if it works now :)

Topazio29, 2024-06-09:
I solved the problem by running the PID for both the hotend and the bed at their respective working temperatures. I no longer encountered the problem that occurred quite frequently. Thanks for the great suggestion, Oliver.

closed 2024-06-10 by knutwurst