knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #541: [BUG]

dimar01, 2024-07-03:
### Bug Description

After installing the new firmware (DGUS_TMC_v1.5.3) as well as TMC2209 drivers in my Mega X, the z-axis makes a lot of noise compared to the x- and y-axis

### Steps to Reproduce

1. Install TMC 2209
2. Install firmware (MEGA_X_DGUS_TMC_v1.5.3)

**Expected behavior:** silent axes

**Actual behavior:** noisy z axis

#### Additional Information
I'll upload a video to show the problem

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/assets/107865762/a91f9a7d-a356-43c8-bf98-3cf9929d2171

knutwurst, 2024-11-22:
Did you load the firmware defaults after flashing? To be honest: That doesn't sound so bad. It's about the maximum speed that the Z axis can use. I chose it like that so that Z-hops are also possible with the mega x. Since you have installed the TMC2209 driver, the other axes seem a lot quieter.

But you can also limit the maximum speed of Z via gcode. Then the printer is a bit slower, but also quieter.

dimar01, 2024-11-22:
Look at the video it doesn't work but live the noise is really great. I also tried to put the file for the mega s printer and the noise disappears, but it doesn't set me home