knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #546: Control of motor direction

elwaspezo, 2024-08-21:
### Description

Changing the motor direction from the user interface would be simplest, yet powerfull feature. To be precise: I've switched to dual drive extruder, so I have to change the E motor direction. It's no use. 

### Feature Workflow

Additional fetures on UI level would do the job...

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.

stklcode, 2024-08-22:
It’s not really the “simplest“ feature.
Okay, there’s a common hack to set negative number of steps per unit which _should_ work for the extruder when using units/mm (never tried it). Not for other axis though as it does not cover homing, min/max, etc.
Talking about the [`M92`](https://marlinfw.org/docs/gcode/M092.html) command that is also used for extruder calibration `M92 E-...` plus M500 for persistence. 

It’s a thing you typically will use exactly once per printer (or once per EEPROM reset), so axis direction in Marlin is a build-time config for good reasons. I guess build your own firmware once or use M92 via console is an acceptable effort for such one-time action and an additional menu not worth the effort.

elwaspezo, 2024-08-22:
A really quick answer. Thanks!

To be perfectly honest, I thought about negative values, but ... with great
concern ("will it 'budge'? Most likely no..."). Now it appears, that it's
definetly worth of try. Thanks a lot.
As the problem has a common nature, and If the solution is usefull, maybe
it's a good idea to place a note in the "FAQ"?
(Although it's usefull, reversing the motor plug should be considered as a
brute-force practice :) )

elwaspezo, 2024-08-22:
So far, it's no use...

stklcode, 2024-08-23:
Is it really only the direction you need to change, no other parameter (federate, retracts, etc.)?

You can either enable `#define KNUTWURST_BMG` which inverts the direction and sets some parameters for BMC (clone) extruders or modify only this section so specify `INVERT_E0_DIR`
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/c41a85354ac5d83071fe3262d72af7d5c3209b99/Marlin/Configuration.h#L2029-L2033

No idea about your hardware, but my point still holds: an inversion toggle is neither simple nor useful for more than one or two people modifying their hardware. So you’re better off learning how to build the firmware ([Wiki](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Compile-Firmware-with-PlatformIO)) and make the changes as required yourself, once in the code.

elwaspezo, 2024-08-23:
I'm not asking anymore for extra features, just have mentioned, that interesting hack does not work.

I'm skilled enought, to build/compile/recompile AVR software (skilled enough to reimplement AVR in Programable Logic Device, to be honest), but I don't like to modify someone's code  without asking the author - and that's what i did.

Thanks a lot, I know what to do.