knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #551: [BUG] file list shows shortened filename over the entire list

Tobim6, 2024-09-08:
### Bug Description
The list of files randomly shows a shortened filename of one of the files over the entire list (original filename: grill knob_bla_bla_bla, shortened: grillk~1.gco). The filename it uses always seems to be the file I last put on the sd card. To fix it you need to reinsert the sd card

### Steps to Reproduce
1. Put a gcode file on the sd card
2. Put the sd card in the printer
3. Open print menu

**Expected behavior:**
The screen shows the list of the files on the sd card normally
**Actual behavior:**
The screen repeats a single shortened filename through the entire list
## Additional Information
This only happens sometimes.
Link to video:
[https://hovno.yummypear.cz/20240908_205129.mp4](https://hovno.yummypear.cz/20240908_205129.mp4)

HaGueRe, 2024-09-15:
It's possibly due to the hyphen in this file name.
[image]

Tobim6, 2024-09-25:
Nevermind. This happens when the SD Card is removed and reinserted without the sensor detecting it (microsd to sd adapter).

closed 2024-09-26 by Tobim6