knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #553: [BUG] Bed does not show up in ESP3D

Tobim6, 2024-09-26:
### Bug Description

Bed does not show up in ESP3D web UI

### Steps to Reproduce

Install esp01 with knutwurst's esp3d firmware

**Expected behavior:**
Both bed and hotend is shown
**Actual behavior:**
Only hotend is visible
#### Additional Information
Also the config in the WEB UI does not persist on reboot.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[image]

Tobim6, 2024-09-26:
Can be fixed by setting enable_bed to true from false in preferences.json

stklcode, 2025-09-19:
Wrong repo? This one contains Marlin firmware for i3 Mega printers, not ESP3D, not even in the wiki. I don't see how anything could be fixed/improved _here_.

Tobim6, 2025-09-19:
> This one contains Marlin firmware for i3 Mega printers, not ESP3D, not even in the wiki. I don't see how anything could be fixed/improved _here_.

The wiki contains instructions how to install ESP3D for these printers.

closed 2026-08-17 by github-actions[bot]