knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #555: [BUG] USB Not working

AndrexYT, 2024-11-14:
### Bug Description

I flashed the printer (Anycubic 4max pro 2.0) and when i connect it to pc it beeps, but pc just says connecting..., idk how to fix that

### Steps to Reproduce

1. Flash firmware
2. Connect to pc

3. Use for example pronterface to connect

**Expected behavior:** Connect

**Actual behavior:** Beep, fast 3d printer lights flash

#### Additional Information

https://github.com/user-attachments/assets/de025771-cf64-4296-b905-94241a77feb3

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2024-11-14:
Is it possible that you have other programs that access the virtual serial interface? This behavior is actually typical of another program repeatedly resetting the connection.

The beeps from the printer always come when the firmware starts.

Sometimes it is just a bad USB cable.

It is also possible that the potential is equalized via the USB cable if both devices (printer and laptop) are connected to different circuits and both are connected to the protective conductor. Try running the laptop on battery power only without any other cables.

AndrexYT, 2024-11-14:
> Is it possible that you have other programs that access the virtual serial interface? This behavior is actually typical of another program repeatedly resetting the connection.
>
> The beeps from the [...]

The cable is okay, the laptop on video was already running on battery. None of programms were opened

BlockListed, 2025-05-28:
I'm having a similar issues. The printer was working fine over USB for a long time (octoprint). Then after switching to another system and back it stopped connection, sometimes outputting garbage. I have tried multiple USB cables and computers.

rawnullbyte, 2025-05-28:
I got a new 3d printer, no issues at all.

closed 2025-05-28 by rawnullbyte