knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #556: [BUG] Speedoption in the axis menu has no effect on Z ON Anycubic Mega Pro

nicokaiser79, 2024-11-22:
### Bug Description

In the axis menu where the individual axes can be moved, the speed of the respective movement can also be set. This works for X and Y, but not for Z. Also, the movements of the Z axis are generally slow. This works perfectly with the original firmware.

### Steps to Reproduce

1. Go to TOOLS
2. Choose axis
3. Set Speed to L => low
4. Push 1.0 for X+ and X- the speed is slow
5. Push 1.0 for Z+ and Z- the speed is slow
6. Set Speed to H => High
7. Push 1.0 for X+ and X- the speed is fast
8. Push 1.0 for Z+ and Z- the speed is slow

**Expected behavior:** [What you expect to happen]
The speed setting should also have an effect on the Z axis. 
**Actual behavior:** [What actually happens]
Currently, the movement of the Z axis is generally slow

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2024-11-22:
Hi @nicokaiser79 , can you please tell me the exact type of printer and also whether you have made any changes to it?

nicokaiser79, 2024-11-22:
Hi @knutwurst i have a anycubic mega pro. I added a TCM2208 driver with reversed plug for compatibility original firmware. So I switch actually between original and current firmware (laser and 3d printing). P.S: at the moment I want to try to complete the laser issue in the current marlin firmware.

nicokaiser79, 2024-11-22:
So i fix it, I take the values from original firmware. In Configuration.h for KNUTWURST_MEGA_P
#define DEFAULT_AXIS_STEPS_PER_UNIT {80, 80, 400, 384}
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 10, 80 }
#define DEFAULT_MAX_ACCELERATION {1500, 1500, 800, 10000}

closed 2025-10-19 by github-actions[bot]