knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #559: [BUG] Anycubic P - print stops, target temperatures show 0

fenugrec, 2024-12-03:
Hi,
I've installed this firmware hoping it would cure some glitches in the original firmware.

I've had this issue a few times now randomly, not every time. Can happen within 1h. 
 - Print stops at some arbitrary height,
 - parks the head at the origin (nice)
 - the temperature targets now display 0 for both E0 and Bed, as if it had given up.
 - Display is still normal, responsive progress screen. The temperature readings still update, too. 

Is there a way to find out what exactly it's waiting for ?

### Steps to Reproduce
1. print anything
2. cross fingers to make it through the print
3. it doesn't

#### Additional Information

* Anycubic mega P
* this firmare, `MEGA_P_DGUS_v1.5.4`
[image]

fenugrec, 2024-12-05:
I looked at the Gcode, and there are  
- the expected G0, G1 moves
- some M205 X Y  motion settings
- G92 E0 "New extruder position" , a few of these in the early parts to clear filament etc, and one around layer 60, Z=19mm . Unsure what Z height my prints stopped but it seemed quite lower than 19mm.

There is no weird M command in the middle of nowhere, it's all just regular stuff...

fenugrec, 2024-12-06:
Out of pure irrational optimism I tried the print again today, and after failing to even start a few times in a row, it eventually worked. Not sure what to think of this, but it will not be possible to debug this issue further.

closed 2024-12-06 by fenugrec

stklcode, 2024-12-06:
Out of curiosity: printing from SD card or via USB?

I experienced some more or less random filament runout issues (most likely a hardware issue, he sensors are not very reliable) which got better with increased error threshold (since 1.5.3 IIRC) and there was some change around runout pause in 1.5.4, too.

I didn't use the SD card for quite a while, so wondering if it's actually a simple false-positive runout with lack of notification on the display :thinking:

fenugrec, 2024-12-06:
> Out of curiosity: printing from SD card or via USB?

From SD. I tried Cura->USB->print a few times and it failed miserably every time

> I experienced some more or less random filament runout issues

Hmm interesting, I'm only aware of the limitswitch that just senses filament absence, but there would be no threshold for that, are you referring to something else ?

Frankly I'm tired of this printer's unreliability. I'm not sure it's entirely the firmware's fault (whether anycubic's or this build), but the lack of error reporting / logging makes it impossible to tell.