knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #560: [BUG] Hotend Temperature stuck

eigencreation, 2024-12-15:
**Expected behavior:** [What you expect to happen]

When you turn on the printer, the temperature should fluctuate slightly at 21.

**Actual behavior:** [What actually happens]

The hotend stays at 51 and shows no changes. Even if I disconnect the hotend cable, the temperature remains at 51 and does not show the usual -14.

#### Additional Information

Steps I have already completed:
- re-flashing firmware 1.5.4
- Firmware reset
- hotend swapped
- thermisator replaced
- hotend microcontroller cleaned with alcohol
- hotend cable connectors cleaned with alcohol

stklcode, 2024-12-15:
Did it work properly at some point and if yes, what has changed since then (e.g. firmware update or switch)?

Temperature on the display, via serial console or both?

(narrow down the problem to measurement or display communicaiton)

>  Even if I disconnect the hotend cable

Does it make any difference whether you disconnect the cable on the printer side or the hotend side?

(eliminate issues with the wiring harness)

Might be a hardware issue like some broken insulation or short circuit. Stable 51°C should be around 35kΩ NTC resistance ... not very like, but he test is just 10 seconds.

Worst case might be a broken ADC input in the controller. But let's first try to narrow down the issue as much as possible.

eigencreation, 2024-12-18:
After a few researches and measuring on the board, it might be something broken. One part should have 12V, but showing me only 7V. After the repair, i will post again