knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #561: Need BLTouch version of 4MAXP2_DGUS_v1.5.4.hex firmare

dennislia, 2024-12-16:
### Description
My Printer:
- Model 4Max Pro v2 (BMG direct extruder)
- Display 0.0.2 display (DGUS clone)
- Mainboard Trilgorilla 0.0.2 (_DGUS firmware )
- BLTouch sensor V3.1

There isn't a firmware version with BLT in the name that matches my printer, so I tried 4MAXP2_DGUS_v1.5.4.hex.

### Feature Workflow

1) I successfully installed the 4MAXP2_DGUS_v1.5.4.hex firmware.
2) I was able to perform manual mesh leveling via the printer's LCD Special Menu and save to EPROM.
3) I am able to 3D print using the save mesh values.
4) I was able to install the BLTouch sensor and connect to the control board.
5)when printer is turned ON the sensor's probe extends several times, retracts and the RED LED stays ON.
6) using Octoprint gcode terminal sending M280 P0 S60 or any gcode that should be a valid BLTouch code results in an unrecognized command by the marlin firmware
[image]
.
7) This tells me BLtouch isn't supported by this firmware.
[image]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.

stklcode, 2024-12-16:
Anything wrong with `4MAXP2_DGUS_BLT_10_v1.5.4.hex` from the [Releases](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.5.4) page?

dennislia, 2024-12-16:
Hi ,Thanks for answering.Yes, there is a problem with 4MAXP2_DGUS_BLT_10_v1.5.4.hex.I tried that first, but the printer's LCD Z-home went in the wrong direction.  It sent the bed to the bottom(max Z), instead of bringing it to the top (min Z).   The X-home and Y-home worked correctly.I think the incremental Z movement selection from the Tools menu went in the correct direction, but the Home selection sent Z in the wrong direction.I didn't get to try the M280 commands in 4MAXP2_DGUS_BLT_10_v1.5.4.hex, because I assumed, that with the Z-home going in the wrong direction, it was only for the 10 board.I appreciate your interest and recommendations.Thanks,...Denny

stklcode, 2024-12-16:
Weird 🤔 Both mages are built from the same source code in the same pipeline, so I’d expect both showing the very same issue or none.

Seems like this config line might not be correct here:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/6734173854f7718539f8d0d200ef338660a7e755/Marlin/Configuration.h#L1269-L1273
Changed back in 1.4.0 (https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/d0219c5632bb9e55ce1313c85d6eca6aa6ae6f28).

I could create a custom build and upload it later today, but unfortunately I don’t have a 4MAX and I have no idea whether reverting this might break anything else.

dennislia, 2024-12-16:
Thanks again,I would be very happy to try any update that you might be able to make.BTW, I don't see a build before 1.4.0 that I could try.Thanks,...Denny

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/6734173854f7718539f8d0d200ef338660a7e755/Marlin/Configuration.h#L1269-L1273
Changed back in 1.4.0 (d0219c5).

dennislia, 2024-12-17:
Hi Stefan,

If I can get VS to build your GitHub code I could make a custom build to test on my 4Max Pro 2.0.

I am close, but this is my first experience with VS.
I am getting “ Unknown environment names ‘mega2560’ “ error followed by a list of valid names (see 1st pic), but I don’t know how to fix this in VS.
The side bar has a Makefile icon which opens a dialog (see 2nd pic).  Do I use this?

Any advice you have is appreciated.

Thanks,
Denny
￼

￼

stklcode, 2024-12-17:
I got a question according your wiring: Was there anything connected to the Z+ pin before you connected the BLTouch?

The code indicates that "Z+" (D19) is used for Y endstop and the BLTouch reuses the Z_MIN connector.
That's why `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`, `Z_SAFE_HOMING` and `Z_MIN_ENDSTOP_INVERTING false` are defined.

Regarding the VSCode stuff: no idea, didn't use it for years (I typically just open the project in CLion and run PlatformIO from shell)

`mega2560`  is default in upstream Marlin project, but it does not exist in our config. In this repo it should be all of "our" supported environments

_platformio.ini_
```ini
[platformio]
...
default_envs  =
  MEGA_1G
  ...
  4MAXP2_DGUS_TMC_BLT_11
```

You can always specify one or more specific targets. This command should do the job: (or replace `default_envs = ...` with a single target)
```
pio run --environment 4MAXP2_DGUS_BLT_10
```

dennislia, 2024-12-17:
Hi Stefan,Good detective work ...There is a daughter board plugged on top of the Z+,Z-,Y+,Y-,X+,X- pins on the motherboard (see attached pics ...Are you suggesting that I make connection to the Z- pin instead of the Z+ pin ???See info and pics below...Thanks,...DennyOriginal main board connections Note: daughter board appears to make connections to Z+, Z-, Y+, Y-, X+, and X-Removed Daughter board and adjacent 3 pin connector (12V, GND & SCL connections)Needed a GND pin to connect to ... found it at pin 6 of the  ISP connector BLTouch extension cable(different colors)GND  5V  SIG        GND  ZMINBRN RED YEL       BLK   WHT orig cable ORG YEL GRN      BLU   BLK  extension cableBLTouch 2 pin connector wiringSolder black wire to Z+ pin on daughter board Plug blue wire on pin 6 of ISP(GND)BLTouch 3 pin connector wiring (YEL, ORG, GRN ... 5V, GND, SIG)plug connector onto servo port S1 pins ... confirmed the correct order

dennislia, 2024-12-18:
Retry ... ...Denny

dennislia, 2024-12-18:
Hi Stefan,I don't know why, but I had thought that Zmin was short for Zminimum of the sensor.  That threw me off.Tomorrow I will rewire to the Z- pin and install the 4MAXP2_DGUS_BLT_10_v1.5.4.hex firmware, on my 4Max Pro 2.0 printer.And I will hopefully let you know that it successfully works with the BLTouch sensor.Thanks you,...Denny

dennislia, 2024-12-18:
Hi Stefan,Some great news !After connecting BLTouch Zmin wire to Z- pin on mainboard and installing your 4MAXP2_DGUS_BLT_10_v1.5.4.hex firmware, the hot end now goes in the correct directions when homing using the printer LCD controls ... ya!And the printer responds favorably to the all of the BLTouch M280 commands in Octoprint terminal (see pic) ... ya!I have not run the auto bed leveling procedure from the LCD Special Menu or from Octoprint terminal or Cura gcode, because there are 2 issues to address first...ISSUES:1) HOMING End location: when homing, now the hot end goes to the far left(X=0), then full forward(Y=0) and then to the middle of the bed(X halfway,Y halfway) where it twice exercises the BLTouch probe and stops ... (see video) ... - I don't think this is right ...- I expected the probe to end up at X=0 and Y=0 since that is where home is for the 4Max ...2) Y OFFSET when doing ABL using LCD Special Menu: my BLTouch probe is at -54mm relative to the nozzle on the Y axis ... I think I read somewhere that the firmware default was around 20mm ... I have the probe lined up with the nozzle on the X axis, so I think that's ok ...- I think Y Offset is a problem, since I think the probe will be out of the print area when running auto bed leveling from the LCD Special Menu ... how to handle this???- I think the -54mm Y offset could be added to as gcode in Octoprint's terminal and in Cura's start gcode ... I guess this is OK ...Recommendations?Thanks again,...DennyS10 deployed the probe S90 retracted the probeS120 deployed and retracted the probe 10 timesProbe is still RED  after S60 and S160 commands (according to their data sheet I think this is ok)Nozzle ends up in the middle of the bed when Homing (should be full left and full forward)

stklcode, 2024-12-19:
> HOMING End location: when homing, now the hot end goes to the far left(X=0), then full forward(Y=0) and then to the middle of the bed(X halfway,Y halfway) where it twice exercises the BLTouch probe and stops ... 

That’s intentional. When the BLTouch is used for homing, it might not hit the hotbed at X/Y 0. so `SAFE_HOMING` is enabled and Z homing is performed in the middle of the bed instead.

> (see video)

Which video? Think there’s some issue with mail responses, no formatting at al and no attachments.

> I think I read somewhere that the firmware default was around 20mm ... I have the probe lined up with the nozzle on the X axis, so I think that's ok ...- I think Y Offset is a problem, since I think [...]

Actually it’s X -2 Y -25 Z -0.4

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/6734173854f7718539f8d0d200ef338660a7e755/Marlin/Configuration.h#L1841

Can be overwritten via GCode `M851` ( https://marlinfw.org/docs/gcode/M851.html )

dennislia, 2025-01-30:
Hi Stefan and Oliver,

I am pretty much finished with getting BLTouch with New Special Menu to work and tested on my 4MAX Pro 2.0.

I have created a branch, off of the Fork that I made off your code, and released builds for just the 4MAX BLT versions at https://github.com/dennislia/Marlin-2-0-x-Anycubic-i3-MEGA-S-Fork/tree/new-special-menu . Wiki at https://github.com/dennislia/Marlin-2-0-x-Anycubic-i3-MEGA-S-Fork/wiki .

If you are interested in reviewing and considering adopting the changes, I’d appreciate your comments and recommendations, on how to proceed.

Note: I think that every line that I touched I added “//dsl” so that all my changes could easily be found by Find-In-Files.  
6 files were changed: 

1) platformio.ini (forgot to add "//dsl"), 

2) Configuration_adv.h, 

3) Configuration.h, 

4) Version.h, 

5) anycubic_touchscreen.cpp 

6) anycubic_touchscreen.h.   

I changed a lot of the macro definitions to be consistent with the new screen menu selections text. 

Thanks for your support.

Denny
PS( I couldn't find any other way to communicate with you about this ... ie...no email or messaging vehicle in Github ... hope this is ok)