knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #562: [HELP] BLTouch failing despite no obvious installation flaws

marvellousmarvolo, 2025-01-07:
### Bug Description

I installed an original BLTouch unit closely following the instructions in the wiki and yet I can't get it working.
Up until the second test everything seems fine. The automatic self test on boot (pin moves up and down, red glow) works as expected. But when I launch the auto leveling from the special menu, the pin doesn't extend and the nozzle crashes into the print bed.

### Steps to Reproduce

I routed the servo wires like in the pictures below. The order (+, -, S) looks correct to me.

[image]
[image]

I mounted the signal pin headers on holes G1 (black) and G3 (white) while bridging H3 and H4 ("alternative wiring"). There don't seem to be any obvious shorts or unwanted contacts (even under the plastic pin header frame).

[image]
[image]
[image]

I installed firmware version `MEGA_S_DGUS_TMC_BLT_11_v1.5.4.hex` because my printer is an Anycubic Mega S with DGUS-clone screen, upgraded stepper motors and most probably has a V1.1 board according to the placements of parts and servo port descriptions.
Executing the `M280 P0 S10` command to extend the BLTouch pin does nothing.

[image]

Is there anything obvious I have missed? Anything left to try?
Maybe do the conventional pin header install?

Thanks in advance!

knutwurst, 2025-01-07:
Hi @marvellousmarvolo ,
Are you absolutely sure that you have a Trigorilla 1.1 board? This is clearly stated on the Trigorilla logo on the board. Only if it says 1.1 do you need to use the _11 firmware. In all other cases you need the _10 firmware.

I can't see any other (obvious) error at the moment.

Your printer looks good, by the way :) You've processed everything cleanly. It's also great that you've included photos in the bug ticket. That really helps a lot. :)

Olli

marvellousmarvolo, 2025-01-07:
Hi Olli,
thanks for the compliment! Although I would be happier if the printer worked just as well as it apparently looks... 😅

As for the firmware, I also tried out the _10 version, but everything behaved exactly the same. No pin extension followed by a crash into the build plate.

Sadly, the board version marker is covered by a cooling assembly that was already installed, when I got the printer. If really necessary I can take it out and double check but there were some dead giveaways that this is infact V1.1, like the servo ports (PS, S1, S2), the smaller wire clamps for power and the top of the copyright symbol peeking out.

[image]
[image]
[image]

Could the BLTouch itself be the issue? I ordered [this one](https://de.aliexpress.com/item/1005006210601045.html) which appears to be v3.1.

stklcode, 2025-01-07:
From the position of the T in the triangle is could be a 1.1 or a 0.0.3. But it looks like it’s just one screw to uncover the version number. Or simply flash a `_10` build, if it works, good, otherwise it wont break anything, just switch back. Both faster than guessing around.

marvellousmarvolo, 2025-01-07:
> From the position of the T in the triangle is could be a 1.1 or a 0.0.3. But it looks like it’s just one screw to uncover the version number. Or simply flash a `_10` build, if it works, good, otherwise it wont break anything, just switch back. Both faster than guessing around.

Thanks for your input!
Unfortunately, as I have mentioned in my previous comment, I already tried flashing a `_10` firmware with absolutely no difference in behaviour.

marvellousmarvolo, 2025-01-07:
Ok, so to clear up the mystery once and for all, I removed the cooling assembly to reveal the board logo. It is indeed a rare V1.1.

[image]

I also tried out the firmwares 1.4.8 and 1.3.1 (latest minor versions each), but still no difference. The BLTouch reacts to neither the auto leveling nor to commands entered via pronterface.

I don't know whether this is relevant or not, but I also noticed, that the red LED switches off when I manually pull out the probe. In some videos I have seen that instead a blue LED should turn on, is that correct?

[image]
[image]

marvellousmarvolo, 2025-01-10:
@knutwurst could it be that the servo pinout is somehow incorrect? According to [this diagram](https://www.thingiverse.com/thing:3680634) of the V1.1 board the pin order is `11 4 6 5` instead of `5 4 11 6`.

stklcode, 2025-01-11:
Rightmost in the diagram is S0, so `S0 S1 S2 S3` is `11 6 5 4` for 1.0 and `5 6 4 11` for 1.1

That does not match, but in either case `S0 = 5` is used for the BLT on 1.1

Hard to tell if that’s incorrect. You could measure on you board or build FW versions with BLT on Servo 1/2/3 to find out which pin is actually attached...

Order in fir FW (upstream Marlin) was never changed, guess someone might have noticed if Servo 0 was incorrect (1-3 are rarely used, those might be wrong) 

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/6734173854f7718539f8d0d200ef338660a7e755/Marlin/src/pins/ramps/pins_TRIGORILLA_14.h#L34-L39
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/6734173854f7718539f8d0d200ef338660a7e755/Marlin/src/pins/ramps/pins_RAMPS.h#L65-L80