knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #567: [HELP] TMC2209 correct Vref, no torque, skipping lots of steps

bisak, 2025-01-30:
Hey, I realise this is not a support forum, but since I'm unsure if this is a software or a hardware issue, or perhaps a combination of both, I'll just ask.

I bought[ these TMC2209 drivers](https://www.aliexpress.com/item/1005006196489026.html), flashed the correct _TMC firmware for my machine, coming from the non-TMC one. Everything moves in the correct direction, but when moving the motors via the menu on H speed, especially the Y axis, it skips a lot of steps. 

Printing is pretty much impossible because the Y axis skips tens of steps on every travel move.

I've tried various Vrefs in the range of 1 to 1.4v with no change. I've also swapped around the drivers with no change. The stock drivers still work great.

Could this be some configuration that I'm missing? Thanks for any input.

knutwurst, 2025-01-30:
Hi :) So with TMC drivers it depends a lot on the axis that is to be moved. I think Vref of 1.4V is too high, as the motor current increases too much here. The Y-axis can definitely handle 1.2V, though. 1V on the other axes can also be OK.

Cooling is extremely important! You have to cool the drivers properly with a fan because they get much hotter than the stock ones. No fan ducts. No 120mm fans that don't generate any airflow in the middle, just normal, clean cooling with one or two small fans.

Then it is also extremely important that you load the defaults after flashing, as your acceleration values ​​may be much too high and the drivers or motors simply cannot cope. The stock drivers have more power even when uncooled. That's just the way it is. But StealthChop is missing.

Finally, if I were you, I would swap the TMC drivers around. So X against Y or against Z. Just to see if one of the drivers is defective. This can always happen with Bigtreetech.

I hope this helps :)

bisak, 2025-01-30:
Thank you for the super speedy reply.

I'm using the stock fan mount with the stock fan size for the drivers, I've just upgraded it a more quiet and powerful Sunon Maglev.
Since this happens on the very very first move too, I doubt it's heat since the drivers must be stone cold at that point. Im running ~6.5 jerk and ~1600 acceleration

I've tried swapping the drivers around, no change. All motors lack torque, even the extruder motor skips steps when using the TMCs, compared to the stock ones with the same temperature and same filament.

Is it even possible that I get 5 defective TMCs, can they be such bad quality that none of them works properly?