knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #569: [HELP NEEDED] i3 Mega with dual endstops. Are the endstops still used?

RebelSoftware, 2025-02-09:
I have installed a 3d touch and used your marvelous instructions to get it all working. but I haven't done any printing yet because I wanted to find out about the original end stops.

I have looked in the FAQ on your wiki and searched the source code but unfortunately it was not clear to me what the answer is.

Sorry if my question is bleedingly obvious and I have just missed it due to my lack of understanding.

My question relates the whether after running g29 and storing the bed leveling information, the original physical (dual) end stops are still used?

The reason why this has come up is if I do a bed leveling pass and then pick a random spot on the bed and send the print head to it using say "g90", "g1 x100 y100" and then "g1 z0", the paper test does not quite work. When setting the original height and z-offset for the nozzle the paper grabs the print head when moved side to side just a little, however when I move the head to a random position near the middle of the bed and do the same test (with the bed heated) the paper slides freely.

So I was wondering if the original z end stops are overriding the head position, perhaps preventing the head from going lower?

I did multiple bed levels to make sure that the 3d touch was relatively consistent and it seems to be ok with a max variation of around 7 hundredths of a millimetre, I am guessing that's acceptable.

I also have a feature request and thought I would float it here. I saw a bed leveling video for Geeetech where the probe tested twice at each location. I presume the average of the probe position was then calculated and used.

I wondered if the same could be done with your firmware as it would counter minor sensor variation like I had above.

Great work by the way I appreciate what you have done and are doing.

stklcode, 2025-02-09:
It depends.

Normally, physical endstops are only used for homing, i.e. get the absolute position within the machine frame once. (can be enabled always with [M120](https://marlinfw.org/docs/gcode/M120.html), but not by default) Once the printer knows its position, it uses soft-endstops, so the software prevents moving below 0 or above max on any axis (can be controlled with [M211](https://marlinfw.org/docs/gcode/M211.html), e.g. when doing offset calibration).

Soft-endstops take configured/measured offsets, when defined.

RebelSoftware, 2025-02-09:
"Normally, physical endstops are only used for homing," Great this was what I was expecting. it also ensures the tramming is correct,  as the i3 mega needs both end stops to ensure both z-axis motors are aligned.  I assume from what you are saying is that they are then ignored during print and only the soft endstops are used.

However one issue I can see is that on the i3 mega the endstops are physical. in other words, apart from a little flex in the lever of the microswitch the mount of the endstop physically stops the movement on the z axiz.

I had considered changing the microswitches to either reed switches or optical switches which then don't act as a physical barrier and are more reliably triggered.

stklcode, 2025-02-10:
True. They are mechanical switches and you may even break or misalign them, if you move too bar below 0. But, if you really need a level <-1mm or somewhere in that area which should be tolerated with the switches, there is likely something else wrong with the printer or the bed. Typical misalignment should be well below, so the switches are just "good enough" for that purpose.

Nevertheless, feel free to swap them, noone's gonna stop you from "optimizing" anything.

bgolob, 2025-08-25:
With this firmware, can I remove both z endstops, or do I have to keep them on?. At the moment I have them on but Im not sure how to set everything up. Do i have to tighten the screws all the way and then manually set the z-endstops so the nozle touches the bed and then configure bl touch?