knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #570: Bl Touch bringt falsche Werte

wesleym010904, 2025-02-18:
Nach dem automatischen Leveln mit dem bl Touch und den richtig eingestellten z- Offset erscheint bei mir das Fehlerbild dass es den Düsenabstand nach rechts zum Druckbett immer weiter entfernt .
Links erziehlt man ein hervorragendes Druckbild auf dem First layer. 
In der Mitte sieht man schon einzelne Bahnen und Recht ist der Düsenabstand so groß daß es nicht einmal mehr für eine Haftung am Druckbett reicht woran liegt dieser Fehler

airdra, 2025-02-18:
exakt dasselbe Phänomen wie bei mir, leider habe ich bisher auch keine Lösung dazu.

knutwurst, 2025-02-19:
Das Problem ist bekannt und in den FAQ beschrieben: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(deutsch)#ich-habe-in-meinem-i3-megamega-smega-pro-einen-bltouch-sensor-eingebaut-doch-eine-seite-des-druckbettes-hat-einen-höheren-abstand-zur-düse-als-die-andere-das-betrifft-aber-nur-den-unterschied-zwischen-der-linken-und-der-rechten-seite

wesleym010904, 2025-02-19:
Ouh besten Dank ich dachte ich habe schon alles möglich darüber gelesen aber offensichtlich habe ich das übersehen. Das klingt sehr gut ich werde es probieren und hoffen dass es klappt 😅 besten Dank erstmal

airdra, 2025-02-19:
Allerbesten Dank für den Hinweis!! Hat bei mir funktioniert.

Toby2792, 2025-12-10:
Hallo zusammen, es scheint als habe ich genau das gleiche Problem bei mir.  Kann mir jemand die Erklärung zu Verfügung stellen, oder hier beschreiben?  Der Link zu Hansis Block scheint nicht mehr zu funktionieren.   Danke

closed 2026-11-06 by github-actions[bot]