knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #572: [BUG] Anycubic Mega X freezes from time to time

Crady123, 2025-02-27:
### Bug Description

When loading / unloading filament in printers menu - or just moving axis in printers menu my Mega X sometimes freezes.
It does not react to any touches also the temperature shown does not react. It shows 230 degree e.g. but the nozzle is cold.
When switching the printer on / off it works again.

### Steps already done
I bought it as used one - with already KW 1.5.2 was installed.
It already had this behavior with this Firmware.
Now I updated to KW 1.5.4 - still same issue.
File taken:  MEGA_X_TMC_BLT_10_v1.5.4.hex 
Its a 0.0.2 board, with BL-Touch (or simular) and since it works in the correct direction, TMC should be build in too.

stklcode, 2025-02-27:
Does the printer still react to commands via USB?

Question is whether the display itself freezes, the mainboard/firmware or the serial communication between both.

Could be anything from a firmware bug (which does not trigger the watchdog 🤔), to a broken display or simply a bad connector on the display cable. So let’s try to narrow it down.

VodkaCorpsman, 2025-03-24:
Yes I have had this exact same issue, I posted a while ago but no answer, so I've just switched back to the standard firmware