knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #573: [HELP NEEDED] Mega S, BL TOUCH does not extend when z homing

blairjanis, 2025-03-01:
Mega S with clone DGUS, 1.0 board, and BL Touch.  This setup worked on an older version of firmware from a few years ago when originally installing the BL Touch.  I recently applied the AI3M MK4 assembly and wanted to also upgrade firmware.  What I've done:

- Downloaded and installed MEGA_S_DGUS_BLT_10_v1.5.4.hex.  
- Loaded FW Defaults and saved to EEPROM.  

Everything looks good and seems functional except that when I home or start leveling the BL Touch does not extend.  When I start the machine the BL Touch does do two probe actions as expected.  At this point I'm unable to run leveling and adjust any offsets.

I've triple-checked the wiring of the z-stop wires in the BL Touch.  Like I said, this worked with an older version of firmware.  Are there steps I'm missing in the process above?  Could there be a problem in the firmware as it relates to this setup that I could look at?  I've scoured settings and don't see anything amiss but firmware is not my specialty.

Thanks for any help anyone may be able to provide.

stklcode, 2025-03-01:
Why should the probe extend while homing? It’s not use at this point and might even break the probe when the bed is heavily misaligned 🤔 

Did you run a modified build with probe as endstop previously? Otherwise this is expected behavior. I frankly can’t remember if it was different in any old version, but if I had to answer I’d say no. Probe should extend during startup, manually using `M280 P0 S10` (or `S160` to reset) and when leveling with `G29`.

blairjanis, 2025-03-01:
Yes, in my previous firmware I had the probe acting as the Z stop.  I've disconnected the wiring for the z-stop sensors.  It appears I've made an incorrect assumption and am realizing that my setup is different than most.  I'll see if I can go to the source and find the lines that make that switch in the firmware.  Thanks for the quick reply!

blairjanis, 2025-03-01:
Thanks @stklcode!!  You prompted me in the right direction and it's working now.  If anyone else is using the probe for the z-stop on the Mega S there are three modifications needed to the firmware code:

1. `Configuration.h` ... uncomment line 1630 to force the use of the probe for Z-axis homing

```
// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING
```

2. `Configuration.h` ... comment out lines `2715` and `2717` to enable `Z_SAFE_HOMING`
```
// #if BOTH(KNUTWURST_BLTOUCH, KNUTWURST_4MAXP2)
  #define Z_SAFE_HOMING
// #endif
```

3. `Configuration_adv.h` ... comment out line `871` to disable `Z_MULTI_ENDSTOPS`

```
#if DISABLED(KNUTWURST_ONE_Z_ENDSTOP)
  // #define Z_MULTI_ENDSTOPS          // Other Z axes have their own endstops
#endif
```

I'm not sure if this warrants another pre-configured build option.  I'd be happy to try to add the necessary configuration changes for a new build in the code and put in a pull request but I'm guessing that most people aren't using this option since the issue hasn't been raised before.  Or maybe most people are smarter than me and don't make assumptions.  :-)

Thanks again for the help.