knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #574: [BUG] PID overriden when deltaT over 10C

mp-pomorszcz, 2025-03-09:
### Bug Description

When temperature is being set, the PID result is overwritten when distance from target is greater then 10C. In some cases this makes it impossible to tune the PID via autotune, as there is a huge step when leaving the 10C range.
### Steps to Reproduce

1. Set PID to only linear value, for example 10.
2. Set some temperature (for example 135C
3. The resulting heating is:
Recv:  T:123.00 /135.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:124.59 /135.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:126.49 /135.00 B:24.30 /0.00 @:42 B@:0
Recv:  T:128.57 /135.00 B:24.30 /0.00 @:32 B@:0
Recv:  T:130.52 /135.00 B:24.26 /0.00 @:22 B@:0

The '@' value is PWM setting in range [0:127]. This seems to be actual PID value divided by two.

**Expected behavior:** [What you expect to happen]
The PWM value should be continuous, otherwise tuning is next to impossible in some cases.

**Actual behavior:** [What actually happens]

For some unknown reason there seems to be ovverride at delta T 10C that forces heating element to go full power or 0 power. Examples with PID set as M301 P10 I0 D0:

from below: 
Recv:  T:123.00 /135.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:124.59 /135.00 B:24.22 /0.00 @:127 B@:0
Recv:  T:126.49 /135.00 B:24.30 /0.00 @:42 B@:0
Recv:  T:128.57 /135.00 B:24.30 /0.00 @:32 B@:0
Recv:  T:130.52 /135.00 B:24.26 /0.00 @:22 B@:0

from above:
Recv:  T:144.98 /135.00 B:24.80 /0.00 @:0 B@:0
Recv:  T:143.37 /135.00 B:24.65 /0.00 @:0 B@:0
Recv:  T:141.63 /135.00 B:24.80 /0.00 @:0 B@:0
Recv:  T:140.00 /135.00 B:24.77 /0.00 @:0 B@:0
Recv:  T:138.37 /135.00 B:24.77 /0.00 @:0 B@:0
Recv:  T:136.75 /135.00 B:24.73 /0.00 @:0 B@:0
Recv:  T:135.10 /135.00 B:24.69 /0.00 @:0 B@:0
Recv:  T:133.45 /135.00 B:24.73 /0.00 @:7 B@:0

#### Additional Information
I have modified my hotend with different block and heater. This may be more significant when there is lots of thermal mass in the block (or maybe I got a heater element that is lower powered then original somewhat). I started to dig into tuning beyond normal tune because I often had safety timer cut my initial heating. While we are at it - increasing this time by factor of 2 is not a bad idea - at higher temps the losses to the air are about 80% of heater power, so last 10C takes a moment and in default auto tune the timer cuts out...

mp-pomorszcz, 2025-03-10:
Oh... it's not a bug. IT'S A FEATURE.

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/6734173854f7718539f8d0d200ef338660a7e755/Marlin/Configuration.h#L966

With yours default value of 12, wouldn't it make more sense to change it to 25? This way the response of the PMW would be smooth instead of a step at deltaT 10C

stklcode, 2025-03-26:
Try it and report back, if any value different from Marlin default 10 gives you better results. And if so, this can be changed.

There’s no point I guessing here, so somebody has to gather some real-world numbers.

(I assume all relevant parts are stock, we should not "optimize" any default parameters for modified hotends or similar)